esp8266/esp8266.github.io #4: Corrupt file

magtimmermans, 2019-05-11:
Hi,

Travis CI corrupted the json file by adding:

Dropping version 1.6.5-947-g39819f0
Dropping version 2.5.0-beta1
Dropping version 2.5.0-beta2
Dropping version 2.5.0-beta3
Dropping version 2.4.0-rc1
Dropping version 2.4.0-rc2

I also cannot contact him.

jeroenst, 2019-05-11:
I also can't compile my project anymore due to this bug

80avin, 2019-05-11:
I've forked and edited the file to remove corruption.
Use this [link](https://raw.githubusercontent.com/esp8266/esp8266.github.io/8f2dd4e25473dd3b0c80d5d99f28f3ade0b77af6/stable/package_esp8266com_index.json).
https://raw.githubusercontent.com/esp8266/esp8266.github.io/8f2dd4e25473dd3b0c80d5d99f28f3ade0b77af6/stable/package_esp8266com_index.json

magtimmermans, 2019-05-11:
> I've forked and edited the file to remove corruption.
> Use this [link](https://raw.githubusercontent.com/esp8266/esp8266.github.io/8f2dd4e25473dd3b0c80d5d99f28f3ade0b77af6/stable/package_esp8266com_index.json) and choose 2.5.0 version.
> https://raw.githubusercontent.com/esp8266/esp8266.github.io/8f2dd4e25473dd3b0c80d5d99f28f3ade0b77af6/stable/package_esp8266com_index.json

That won't work as the complete file is full with failures.

KingUdo, 2019-05-11:
Same Here! 
Using the previous commit works:

https://raw.githubusercontent.com/esp8266/esp8266.github.io/e17d1956b970cabc34e4005227f35e3dd788f568/stable/package_esp8266com_index.json

okpapt, 2020-04-13:
Hello,
I have same problem with :
https://github.com/esp8266/esp8266.github.io/blob/master/stable/package_esp8266com_index.json
or
https://github.com/esp8266/Arduino/releases/download/2.6.3/esp8266-2.6.3.zip
Can you say me how can I do it well ?
Thank you.

closed 2021-08-03 by earlephilhower